dudanov/iot-uni-dongle #7: Does the dongle also work with ESP32-C3-12F module?

itanczos, 2022-04-25:
Does the dongle also work with ESP32-C3-12F module?
In principle, they are PIN compatible.
Did anyone try?
[image]

portiktamas, 2022-05-03:
Here is a website comparing the different esp8266 modules. The 12F is better than the 12E, because of better antenna thus better reception. https://aruneworld.com/embedded/esp8266/esp8266-boards/

itanczos, 2022-05-03:
> Here is a website comparing the different esp8266 modules. The 12F is better than the 12E, because of better antenna thus better reception. https://aruneworld.com/embedded/esp8266/esp8266-boards/

It wasn't the ESP8266 chip that was the issue, it was ESP32.

portiktamas, 2022-05-04:
Why do you want to use esp32? The communication can be achieved with the esp8266. It's a waste of resources and money

guerocp, 2022-10-25:
> Does the dongle also work with ESP32-C3-12F module? In principle, they are PIN compatible. Did anyone try? [image]

have you tried with the esp-c3-12f , module?

rrelande, 2022-12-09:
thanks for this suggestion. in principle it should be supported. There are some highly interesting features available only in ESP32. 
Despite being claimed as pin-to-pin compatible, there are a few minor differences. it is especially in bootstrapping. 

for here you can read about the points to consider
https://www.esp32.com/viewtopic.php?t=24511

rrelande, 2022-12-13:
bad, bad news regarding the very promising ESP-C3-12F. 
It is officially discontinued by Ai Thinker. 

They have a replacement chip which still respect the ESP-12F pins. 
but it is based on BL-602 chip (Boufalo Labs), which has nothing to do with Espressif. 

I am not sure you can reach the targets set for ESP32 with a BL602