dudanov/iot-uni-dongle #12: Building problem: CPL parts Q1 and Q2 90 degrees out of alignment with trace contacts

snetsplit, 2022-07-25:
Two of the parts, Q1 and Q2 are 90 degrees rotated out of alignment with the contacts. I had to modify the cpl.csv to rotate them, but there's two possible rotations. There's a purple dot in JLCPCB order tool but I'm not sure which side it goes on. From the animation it looks like the label text should go down to up. So I hope I'm correct, but I don't know.

Can someone help

How the CPL file is off here
[image]

Is this how it should be?
[image]

malves61, 2022-07-26:
I have ordered from jlcpcb a few months ago, 5 boards were manufactured as your 'best guess' without any modification. For what is worth I have not yet finished assembling them (ESP8266 and USB connector) or tested them - you know 'projects' piled over it.

snetsplit, 2022-07-26:
Gonna just hope and order. If it's wrong should be able to resolder it.

ozumado, 2022-08-18:
Hi, did you find out whats the right direction?

snetsplit, 2022-08-20:
Nope, and won't for a bit. Unexpected vet bill and messed up my finances for a bit

ozumado, 2022-08-20:
So I just came with a brilliant idea. I have purchased the iot-uni-dongle last year and its in use at my parents house. I will take high resolution/quality pictures so it will be easier to verify the correct orientation.

snetsplit, 2022-08-21:
That would be awesome!

dudanov, 2022-08-22:
Guys, the rotation angles of the components are not standardized. Whichever angle you specify, there is no guarantee that it will be correct. Usually, in PCB assembly services, this stage is manually pre-moderated by engineers. In JLCPCB so, in PCBWAY too. As a general rule, it is recommended to use a silkscreen that identifies the first pin of a component. Engineers basically rely on it.

Don't waste your time looking for the right rotation angle. Trust PCB assembly service engineers. 😉

snetsplit, 2022-08-27:
Very good to know! Thank you!