dudanov/iot-uni-dongle #19: Tindie Shop Link Broken

SJensen721, 2023-07-07:
Each time I click on the Tindie shop link, it says there is some sort of breakage issue. I can't buy any of these because I'm unable to see the store in Tindie.

iMrLopez, 2024-03-30:
I was able to open the store 3 days ago but it goes back to a 404 again

XanderStrike, 2024-05-30:
Interestingly it appears to show up in search results, but clicking the link takes me to an error page. I've tried both logged in and anonymous.

[image]

@dudanov is the dongle supposed to still be fore sale?

iMrLopez, 2024-05-30:
@XanderStrike  i went all the way on contacting him through Telegram but got no answer at all, i ended up making my own pcbs with the schematics but didnt get to setting them up yet, i am planning on selling them in the future when i get around setting them up

samepant, 2024-06-26:
@iMrLopez did you ever end up selling these? i'd buy a couple

XanderStrike, 2024-06-27:
FWIW I bought [the alternative dongle](https://www.tindie.com/products/smartlightme/wifi-dongle-for-air-conditioners-midea-electrolux/) I mentioned in my previous comment and it works awesome.

dudanov, 2024-07-03:
Hi. I apologize for the long response and for not responding on Telegram. I do not sell modules from February 24, 2022. Not an easy situation. I didn't think things would turn out this way.

**You can create and sell modules yourself using Gerber project files.**

With this message I want to put an end to this issue.

closed 2024-07-03 by dudanov