dudanov/iot-uni-dongle #22: Buck Converter

najdanovicivan, 2024-01-17:
I have AUX Climate which originally uses BroadLink WT1HB12 module. 

AC provides only 12V to the module not 5V which is to much voltage for the LDO

The orignial module has 5 PINS 

TX
RX
EN - Enable pin for step down converter
GND
VDD - 12V when EN is HIGH

Can we make updated version of the dongle which will have option to use step down based on something MP2307 or MP2315 for smaller footprint instead of LDO ?

barszczuPro, 2024-07-01:
I'll ask again because I'm also interested in this topic

barszczuPro, 2024-07-26:
Hello everyone 👋🏻 I'm back with an update... 

I had the same problem that my Aux to USB had 16V, so I designed such an adapter - Buck Converter. And everything works great! 🔥🔥🔥

What do we need:

- Buck converter board from @barszczuPro
- [5V 1A STEP-DOWN D24V10F5 POLOLU](https://www.tme.eu/pl/details/pololu-2831/przetwornice/pololu/5v-1a-step-down-d24v10f5/)
- USB A THT socket
- USB A THT plug

Below are some photos of what it looks like, let me know if you are interested in the files so that you can order such tiles?

[image]
[image]
[image]
[image]

@dudanov What do you think about this additional adapter? Maybe it's worth adding this to your documentation?

the-mentor, 2024-08-25:
@barszczuPro do you mind sharing your board design ?